Digi-Key/digikey-kicad-library #7: How to handle parts that go obsolete or nonstock?

bombledmonk, 2017-12-14:
I'm soliciting anyone's opinion.

Any stocking information will be updated in the fields, but a user selecting the symbols won't necessarily have a good idea that something is no longer stocking or readily available on the website.  The current track of the library is that someone can download the library and they can count on being able to count on it to be normally stocking.  Does this sound reasonable?  How might you handle it?

Should these be moved to separate libraries?  
Should names be appended to alert people? 
Should these be left in place?

domhardt, 2017-12-26:
When developing software the use of an old API call will result in a "deprecated" compiler warning. According to this I would suggest a method where the use of obsolete or nonstock parts lead to a warning. The DRC would be the wrong place for this. Maybe there is a possibility for a warning when generating the BOM.

bombledmonk, 2018-01-29:
The current train of thought will be to append _obosolete _nonstock or _nrnd to the library names so it's less likely that people will use any parts that are not readily available.  Parts will roll off into an obsolete library in after a year or two.  Comments are still welcome

closed 2018-02-15 by bombledmonk