Digi-Key/digikey-kicad-library #14: Remove parenthesis in the library names.

eyob-boye, 2018-02-17:
Remove parenthesis from the names of symbol libraries such as "dk_Coaxial-Connectors-(RF)". The presence of parenthesis give trouble when sym-lib-table is being parsed by kicad.

The convention for what to use is here http://kicad-pcb.org/libraries/klc/

bombledmonk, 2018-02-20:
Shouldn't be a problem to change this, but out of curiosity what kind of problem to does this introduce?  Maybe it's a cross platform issue, but I haven't run across obvious errors related to this.

eyob-boye, 2018-02-21:
You probably wont see an issue on kicad 4.x.x. I am on a nightly build of kicad which contains the new way of registering schematic symbol libraries using sym-lib-table just like the fp-lib-table for footprints that is used today in kicad 4. At least the kicad revision I am testing barfs on parsing the symbol library names containing parenthesis. This is understandable when you see that parenthesis is an important token in the language (Scheme I think...) used to encode these *-lib-table files. I am sure if someone tries hard enough, they can make it work. But why bother, just fix it on your side and all will be well.

bombledmonk, 2018-02-21:
Fixed and pushed.

closed 2018-02-21 by bombledmonk