Digi-Key/digikey-kicad-library #15: Longevity?

gcormier, 2018-03-06:
I'm stoked Digikey took this route, I think it will push the community forward. I use Digikey exclusively for my parts, and would love to use this library for an upcoming open sourced design.

I'm wondering if there are plans to handle longevity - if a part is obsolete or no longer stocked, can we know that a shared schematic will still work without having to overhaul all components in 2 years? What is a reasonable lifespan to expect for a schematic? (Or, even if components are obsolete, stock with those footprints might still exist for larger companies for some time...)

bombledmonk, 2018-03-06:
If you are talking about parts remaining in the library, they will remain for the foreseeable future.  Please see Issue #7 and [this thread](https://forum.kicad.info/t/digi-key-open-sources-alpha-version-of-an-atomic-parts-library/8520/16) on kicad.info for some of the discussion on how we plan on handling that.  It's possible, going into the future (years), some parts may roll off into another obsolete library if there gets to be plenty of bloat, but we won't be removing them completely.

That said, it's always recommended to keep your lib cache with your project as it is quite possible that items in the library will be update and changed over time.

gcormier, 2018-03-07:
Nice, it looks like v5 will help by embedding in the actual project - this will greatly help the longevity and perhaps simplify the library management for you guys :)

closed 2018-03-07 by gcormier

bombledmonk, 2018-03-07:
Actually if I'm understanding correctly, I don't think we'll get embedded symbols until v6.  https://lists.launchpad.net/kicad-developers/msg34707.html