Digi-Key/digikey-kicad-library #18: Transistor symbols wrong

make-ing, 2018-04-11:
Hello,

all symbols in the "dk_Transistors-FETs-MOSFETs-Single.lib" are wrong. The arrow in the symbols should allways be connected to the source pin, but it is always connected to drain. I'm not sure if this is also afecting other librarys but in this one all symbols are wrong.

Also sometimes the diode is in the wrong direction.

If i can find the time i will make a pull request with the fixed lib.

bombledmonk, 2018-04-11:
We're on it, no need for a pull request.

make-ing, 2018-04-12:
thanks for your great work!

make-ing, 2018-06-05:
I think the commit 2c04ab4b9b928dc4527df5f2b0648b537f7abac6 fixed the issue. I didn't checked all but most. I will close this issue.

closed 2018-06-05 by make-ing