Digi-Key/digikey-kicad-library #61: USB symbol pins missing functions in latest update

aklofas, 2018-06-13:
The latest update added better symbol pictures for most of the USB connectors (great!). However, the pins are now missing their functions. Is it possible to add back the labels for VUSB, D+, D-, etc? Those really helped prevent errors.

[image]

Reid-L, 2018-06-14:
Thank you for the insight and feedback, we are in the process of taking a look at resolving this.

Thanks,
Reid L.

ARollens, 2018-06-14:
aklofas,

Thank you again for your feedback. I firstly would like to explain the decision to omit the pin names in the recent release was to try and clean up the look. As your graphic shows, the previous iteration has the pin names and the pin numbers overlapping and it looks messy. Since the pinout information for basic connectors like this is available over the internet we thought that this omission would be very minimal. Your comment spurred us to reopen a dialog on this topic. While we may leave the more complex connectors such as the HDMI, DVI, Display Port, and USB 3.0 for now, we have decided we would try to get these labels back in for the different USB 1.0/2.0 connector types. This image is an example of how we think we will achieve this. We would really to hear what your opinion is of this style so that we can make sure we are on the right track with this design.

[image]

closed 2018-07-03 by bombledmonk