Digi-Key/digikey-kicad-library #73: Are these libraries compatable with KiCAD 5.x?

Sonophoto, 2018-08-28:
MEA CULPA: Not sure if this is the right way/place to ask but I'm on github and this question seemed likely to come up anyway.

Perhaps add some info on versioning to the [README.md](https://github.com/digikey/digikey-kicad-library/blob/master/README.md)?

Reid-L, 2018-08-29:
@Sonophoto 

Noli sollicitari: Thanks for the inquiry and the feedback.

As far as we are aware, through our testing, the libraries are compatible with KiCad 5.x. Most libraries created in KiCad 4.0.7 should be compatible anyway.

We will do what we can to get some versioning and compatibility information in the README. Thanks for the feedback.

Thanks,
Reid L.
Digi-Key Electronics

Sonophoto, 2018-09-17:
update: if you do add something to the readme.md this might be a good link to add to it:

https://forum.digikey.com/c/kicad

mariela1522, 2018-09-19:
Hi, how can this library to kicad? im new sorry

jacky4566, 2018-10-25:
Yes I have been using this library in 5.0 no problem. To import:

For the Symbols:
Open the Symbol Library editor>Preferences>Manage Symbol Libraries>Select "Browse Libraries"
Then select all the dk_xxx.lib files.

For the Footprints:
Follow the same as above and select the .pretty folder.

poeschlr, 2018-12-26:
I can assure anybody wondering about this that any library created in KiCad version 4 is fully compatible with KiCad version 5. (This holds true only for libraries! Schematics need to be ported from version 4 to version 5.)

----

The other way round is not guaranteed. KiCad 5 supports more features for both symbols and footprints. If any symbol or footprint within a given library uses any of the new features then the whole library can not be used in version 4.