Digi-Key/digikey-kicad-library #76: footprint "USB_Micro_B_Female_10118193-0001LF" is incorrect

saviocmc, 2018-10-25:
First of all, thank you all contributors for this awesome library!

I found out that the footprint of the [USB connector 10118193-0001LF](https://github.com/digikey/digikey-kicad-library/blob/master/digikey-footprints.pretty/USB_Micro_B_Female_10118193-0001LF.kicad_mod) is incorrect according to [this datasheet](https://www.amphenol-icc.com/media/wysiwyg/files/drawing/10118193.pdf).

The pads for the shell legs are too far apart from each other. The datasheet specifies 6.10 mm and in the footprint the distance is 6.60 mm (I found that out the worst way, in the assembly of my PCB).

Also, the four landing pads of the shell are missing.

The pads for the signal and power pins are OK.

Once again, thank you all for the work in this repo.

Reid-L, 2018-10-25:
@saviocmc 

Thank you for the feedback. Sorry the error had to be caught in the way it did. I have fixed the issues you pointed out.

Hopefully the fix will be brought into the main libraries sometime today.

Thank you,
Reid L.

bombledmonk, 2018-10-26:
This should be fixed, let us know if you find any more issues.

closed 2018-10-26 by bombledmonk