Digi-Key/digikey-kicad-library #84: 12401610E4#2A USB-C footprint error

jeffmakes, 2019-02-11:
Hi, 
There is an error in the footprint for the 12401610E4#2A right angle female 24-pin USB-C connector. The slots to accept the mounting tabs are incorrectly placed.
Loving the library in general - keep up the good work!
Cheers
Jeff

Reid-L, 2019-02-11:
@jeffmakes 
Jeff,
Thank you for the feedback. I will dig into it.

Reid

Reid-L, 2019-02-12:
@jeffmakes 
Jeff,
Thanks much for the feedback. This should be fixed on pull request [#85 ](https://github.com/digikey/digikey-kicad-library/pull/85).

It will be available as soon as we can do a build.

Thanks again,
Reid

jeffmakes, 2019-02-12:
Thanks Reid, very efficient! I should have mentioned that, as well as the mounting slots, the indication of the recommended board edge is also in the wrong place - I can't plug a cable in to my board because it fouls on the PCB edge before mating. Luckily I can shave off some plastic to get developing :D

Reid-L, 2019-02-12:
@jeffmakes 
Jeff,
I fixed the board edge as well. I also noticed a slight misalignment of the SMD pads. I hope it will serve you well in future layouts.

Thank,
Reid

bombledmonk, 2019-02-12:
Should be pushed up.  Let us know if you have any issues.  Thanks for reporting this.

closed 2019-02-12 by bombledmonk