Digi-Key/digikey-kicad-library #86: Headphone_Jack_2.5mm_SJ1-3523N.kicad_mod

mxa, 2019-03-08:
These Headphone jacks are for a 3.5mm plug, not a 2.5mm one as the name suggests.
https://www.digikey.de/products/de?keywords=SJ1-3523N

In addition the pads are named wrong, I believe instead of 1,2,3 it should be T,S,R

Reid-L, 2019-03-12:
@mxa 

I fixed the Name to be Headphone_Jack_3.5mm_SJ1-3523N.kicad_mod. The pin numbering was hidden so that all pins should connect correctly, they were just hidden. I made them visible and moved the pin names to the inside of the symbol.

Thank you for your feed back. The updates should be available on the next build cycle. Created pull request [#87](https://github.com/digikey/digikey-kicad-library/pull/87)

Reid L.

bombledmonk, 2019-03-21:
This should be fixed as of a few days ago.

closed 2019-03-21 by bombledmonk