Digi-Key/digikey-kicad-library #91: Can't load foot print libraries in KiCAD

Mebuyug7, 2019-03-26:
I am using KiCAD 5.  I found the digi-key-KiCAD libraries and wanted to add them.  I followed a set of directions and was able to load the symbols library was could not load the footprint libraries.  I received an access denied error.  What do I do?

Bob

bombledmonk, 2019-03-27:
What OS are you using?
Where are you saving the files?
Can you include some screenshots of the errors?

Joel-Mckay, 2020-06-04:
I ran into a similar issue with the legacy distro kicad version:
4.0.7+e2-6376~58~ubuntu16.04.1

So upgraded the bin back port:
  sudo add-apt-repository ppa:js-reynaud/kicad-5
  sudo apt-get update
  sudo apt-get install kicad
  sudo apt-get dist-upgrade

Tested lib on: 5.0.2-stable-201811280958+bee76a0~70~ubuntu16.04.1
and it fixed the issue I saw.

As a side note, am I the only one who thinks ghosting people is incredibly rude, and makes Mebuyug look like an ingrate. ;-)

[image]

Mebuyug7, 2020-06-04:
> From    : Joel[mailto:notifications@github.com]
> Sent    : 6/3/2020 8:51:35 PM
> To      : digikey-kicad-library@noreply.github.com
> Cc      : soaked_site@powerc.net; author@noreply.github.com

We are currently on the road, and I will look deeply into what you said later.

As a side note, what the hell does ghosting mean?????  I don't intentionally blow anybody off.

Without further information, I consider you the ingrate.
Bob Shaw