Digi-Key/digikey-kicad-library #98: DK_Detail_Page is not a complete hyper link (also, it is not clickable)

jensenr30, 2019-05-01:
## Forward (thanks!)
The digikey-kicad-library is proving to be a very useful resource in my design process; it's definitely accomplishing Digi-Key's goal of roping more engineers into using Digi-Key as a vendor.  The atomic part paradigm is very useful and saves me time.  Thank you for putting this library together and supporting KiCad (the best free open source CAD program for schematics and PCBs).

## My Issue
For some (all?) parts in the digikey-kicad-library, the **DK_Detail_Page** property is not a valid hyperlink.  It appears to be a fragment of a link.  See the attached image.
[image]

## Proposed Solution
It seems to me that this property simply needs the following string prepended to it: "https://www.digikey.com"
That should make it a valid link.

## Bonus Points
It would also be great if both properties **DK_Datasheet_Link** and **DK_Detail_Page** were *clickable* links.  I wonder if this is something that KiCad supports currently, or if a feature-request would need to be made in order to make user-defined properties able to be made clickable...

Reid-L, 2019-05-02:
@jensenr30 
Ryan,
Thank you for the feedback. I agree that it would be helpful to have the "www.digikey.com" added to the URL. However, this was a conscious decision on our part. The URL is built in that way due to the fact that there are several different sites that could be used by Digi-Key customers. (ex. digikey.ca, digikey.de, and so forth) In order for someone to see pricing in a local currency they would need to be on their local site, therefore we elected to omit the "www.digikey.com" from the URL.

As for the clickable link, we agree 100%. This would be a nice feature. It is however not our call, this would be something that would have to be brought in by the KiCad Developers. I would highly reccomend that you bring it up on the KiCad.info forum under the Feature Requests topic, found here: https://forum.kicad.info/c/community/feature-request 

We will fully consider your feedback going forward.
Thank you.
Reid L

bombledmonk, 2019-05-02:
We were talking about this earlier, and I think there's merit for adding a full URL even if that pushes it into the .com website.  International folk can still easily export boms with PNs and upload it to their respective country's digikey.tld site for purchase.    We'll talk about this a bit more.  

I could swear Alexis made it so any field can now be clickable if it includes a URL, but as I'm checking, that must have been an incorrect memory. I'm going to investigate further.

flummer, 2019-10-20:
What about having a sub domain like redirect.digikey.com prepended, and the first time people visit a URL under that domain, they get to choose the TLD they prefer, and all subsequent hits will be automatically be redirected to the URL on the selected TLD?

Visiting the domain without a path could allow people to change their redirect preference, which would simply be saved in a cookie.

This could also allow anybody to make TLD agnostic links to products, making it easier for international visitors to add the right products to their country specific cart.