Digi-Key/digikey-kicad-library #100: USB_A_Female_Vertical_RA_73725-0110BLF.kicad_mod is backwards

conklech, 2019-05-03:
The courtyard and fab outlines for USB_A_Female_Vertical_RA_73725-0110BLF.kicad_mod are incorrect. Pin 1 is closest to the front of the connector, per the [linked datasheet](http://portal.fciconnect.com/Comergent//fci/drawing/73725.pdf). But the outlines in the footprint extend the other way. Testing with an actual part and PCB (unfortunately...) reveals that the part fits into the footprint backwards. The electrical connection is correct, but the part physically interferes because it extends in the other direction.

[image]

[image]

Either the pads or the outlines should be rotated relative to the other.

Reid-L, 2019-05-16:
@conklech 
Thank you for the feedback. I will get this fixed as soon as possible.

Thanks,
Reid L

Reid-L, 2019-05-16:
@conklech 
I have corrected the issue on pull request #104. This should be available on the next build.

Thanks,
Reid L

closed 2019-07-10 by bombledmonk