Digi-Key/digikey-kicad-library #101: 87180-062LF in dk_Modular-Connectors-Jacks.lib should not include (unpopulated) pins 1,2,5,6

conklech, 2019-05-04:
Per the [linked data sheet](https://www.amphenol-icc.com/media/wysiwyg/files/drawing/87180.pdf) (table at top of page 3) the 87180-062LF variant only has pins 3 and 4 loaded. The other pins should be omitted or, perhaps, have electrical type "not connected" to generate an ERC error if one tries to use them for a connection without noticing that it's a 6P2C part rather than a 6P4C or 6P6C. (Hypothetically.) Compare the [KiCad Library Convention](http://kicad-pcb.org/libraries/klc/S4.5/) on this topic.

Another way to look at it: the included schematic and associated footprint are for the 87180-066LF, not the -062LF. (But that's a more expensive part and probably not suited for many common applications.)

Reid-L, 2019-05-16:
@conklech 
Great catch! I will do what I can to get that fixed right away.
Thanks for the feedback.

Thanks,
Reid L

Reid-L, 2019-05-16:
@conklech 
I have corrected the issue on pull request #104. This should be available on the next build. 
To resolve the issue: 
-I removed the unwanted pins from the symbol 
-I removed the unused pads from the footprint
-I left the pin numbering to be consistent with the data sheet

Thanks,
Reid L

conklech, 2019-05-16:
It might be better to leave through holes or even NC pads on the footprint
to maintain physical compatibility with the corresponding 6p4c or 6p6c
parts, which are otherwise drop in replacements. And it probably would make
sense to add those variants to the library.

That said, thanks for the fixes!

closed 2019-07-10 by bombledmonk