Digi-Key/digikey-kicad-library #103: MMBT3906LT1G symbol: pin-numbering is incorrect.

jensenr30, 2019-05-16:
Summary:
The MMBT3906LT1G symbol in the digikey kicad library has the wrong pin numbers.
The collector should be pin 3, and the emitter should be pin 2.  Currently, they are swapped.  See image below.

As far as I can tell, the reason this this kicad repository exists is to help engineers design faster, and in return Digi-Key will almost certainly make more sales because of the extra value they are delivering to their customers.
A highly detail-oriented person from Digi-Key should verify the parts in this repository are all accurate.  It is not reasonable or fair for Digi-Key to expect the customers to correct these errors in the design/test phase.

This repository is really helpful and it does tend to make me order from Digi-Key over other vendors.  But that becomes less true when the symbols have really simple errors like these.

[image]

Reid-L, 2019-05-16:
@jensenr30 
Ryan,

Sorry for the mistake and thanks for the feedback. I will do my best to fix this as soon as possible.

Thanks,
Reid L

Reid-L, 2019-05-16:
@jensenr30 
Ryan,
Thanks again for the feedback so that we can get this issue corrected. I have corrected the issue on pull request #104. This should be available on the next build.

Thanks,
Reid L

closed 2019-07-10 by bombledmonk