Digi-Key/digikey-kicad-library #119: Footprint Tweak U.FL_Hirose_U.FL-R-SMT-1_Vertical

jacky4566, 2019-07-23:
Just a tweak. The U.FL_Hirose_U.FL-R-SMT-1_Vertical footprint should really be centered on the pin. makes it easier to align with other connectors.

Reid-L, 2019-07-23:
@jacky4566 
Thanks for the feedback. I will take a look at this as soon as possible.

Reid L

Reid-L, 2019-08-01:
@jacky4566 
This footprint is part of the main KiCad library, not the Digi-Key KiCad Library. I would recommend raising an issue on the main library. https://github.com/KiCad/kicad-footprints 

You could potentially use the Coax_Conn_U.FL in the digikey-footprints library, but be ware the pin numbering is different.

Sorry for the delayed response.

Thanks,
Reid L

jacky4566, 2019-08-01:
OH. I should have checked that! thanks.

closed 2019-08-01 by jacky4566