Digi-Key/digikey-kicad-library #120: PCB hole size for OSTTC020162 is too small

goruck, 2019-07-31:
The PCB footprint hole size for OSTTC020162 (dk p/n ED2600-ND) is too small. Its about 0.7 mm but according to the [datasheet](http://www.on-shore.com/wp-content/uploads/OSTTCXX0162.pdf) it should be 1.0mm.

Reid-L, 2019-07-31:
@goruck 
Thank you for the feedback and pull request. The pull request actually changes the footprint to the incorrect pitched connector. I went ahead and correct the whole series of Terminal_Block_xxx as they were all wrong. Pull request #122 should take care of it.

Thanks,
Reid L

goruck, 2019-07-31:
Thanks for addressing this.

closed 2019-07-31 by goruck

goruck, 2019-08-01:
@Reid-L - Quick question. When do the modifications per #122 actually make their way into the digikey-footprints.pretty and digikey-symbol directories? I see the changes made in the src dir. Thanks.

bombledmonk, 2019-08-01:
Sorry, I forgot to push, I'll do it in the morning.

goruck, 2019-08-01:
cool. thx.

bombledmonk, 2019-08-01:
Done.