Digi-Key/digikey-kicad-library #124: HowTo import this?

eabase, 2019-08-01:
How do you actually import this library? 
I'd like to use git clone to ensure my digikey library is always up-to-date.

But there are no instructions on how to import this in such a way that it doesn't get mixed up with the Keycad default libraries. 

Using Windows.

jacky4566, 2019-08-01:
You can add it the same as any other library. They did change the way it works in version 5.
For symbols:
From Eeschema >Preferences > Manage symbol libraries > click the folder icon and browse to folder.
For footprints:
From Pcbnew > Preferences > Manage footprint libraries > click the folder icon and browse to folder.

bombledmonk, 2019-08-01:
I guess it would be prudent to put a link to installation instructions in the readme. I'll work on that.

Reid-L, 2019-08-01:
In the mean time: 

There are several different ways to manage third party libraries like this one. Using git clone is a great way to keep it up to date. 

How you decide to manage the library(s) within KiCad is open to debate, but the way I prefer to use them is to pull them in to the **Project Libraries**, that way they are managed separately from the default libraries, this can be done as such:

1. `git clone https://github.com/Digi-Key/digikey-kicad-library.git` into your preferred directory
2. Open **KiCad**
3. Import the symbols:

    A. Open **Symbol Editor** or **Schematic editor**

    [image]

    B. Click on **Preferences** then on **Manage Symbol Libraries...** 

    [image] 

    C. Click on the **Project Specific Libraries** tab:
 
    [image]
 
    D. Click on the **Add existing library to table** button: 

    [image] 

    E. Navigate to the directory where you cloned the Digi-Key KiCad Library 

    F. Navigate to the **digikey-symbols** folder within **digikey-kicad-library** folder 

    [image] 

    G. At this point you can add one library at a time or all of the libraries. To add all of the libraries click on any  one of the the libraries and then press `Ctrl+a`. This will select all of the libraries, then click the **Open** button.

    H. You should now see the libraries in the table: 

    [image] 

4. Import the footprints(similar process to symbols):
    
   A. Open **Footprint editor** or **PCB Layout**

   [image]

   B. Click on **Preferences** then on **Manage Footprint Libraries...**

   [image]

   C. Click on the **Project Specific Libraries** tab:

   [image]

    D. Click on the **Add existing library to table** button: 

    [image]

    E. Navigate to the directory where you cloned the Digi-Key KiCad Library 

    F. Expand the **digikey-kicad-library** folder and select **digikey-footprints.pretty**

    [image]

    G. Click **OK**

    H. The digikey-footprints should now show in your library table

    [image]

eabase, 2019-08-01:
@Reid-L  
Thanks! *That* should be in the README!

Reid-L, 2019-08-01:
@eabase 
You are welcome! @bombledmonk  added it to the Wiki Page

eabase, 2019-08-01:
Great!

closed 2019-08-01 by eabase

bombledmonk, 2019-08-01:
There's a new section in included in the readme.