Digi-Key/digikey-kicad-library #134: SOT23-8 pinning incorrect

I-Connect, 2020-04-16:
the pins are mirrored, pin 1 should be 8 etc.

It even does not match with the fab and silkscreen pin 1 position marks :-(

jacky4566, 2020-04-16:
Looks like there is some weird stuff with the target part INA219. The pinouts are reveresed for the SOIC and SOT23-8 versions. 

I did a fix real fast on my own branch. 
https://github.com/Digi-Key/digikey-kicad-library/pull/135

You can also just use the kicad version it looks correct.

I-Connect, 2020-04-16:
ina219 has different pinning for different package. 8 pin SOIC is mirrored opposed to the sot32-8 package (found that out the hard way once... :-(  ) So if you use the correct pinning corresponding with the package (as listed in the datasheet) your good.

Except when in this case the SOT23-8 footprint in this library is faulty. Again found that out the hard way. Now I have to see if I can resolder the ina219 upside down.

And yes the footprint in the standard Kicad lib is good. But I expected it in a lib from a respected supplier to be good as well... Lesson for next time, check, check and recheck...