Digi-Key/digikey-kicad-library #138: Digikey KiCad Library is shown invalid

mennyt11, 2020-07-21:
[image]

mennyt11, 2020-07-21:
I'm not able to add the Digikey Kicad footprints even though I successfully added the symbols

bombledmonk, 2020-07-21:
What version of KiCad are you using?

mennyt11, 2020-07-21:
> What version of KiCad are you using?
KiCad 4.0.7

DingoOz, 2021-05-06:
I am in the same boat, using 4.0.7 to follow Shawn's YouTube tutorial and can't load this library.

MizanUrp11, 2023-03-19:
Hi, You can download the library from this repo history  below.
https://codeload.github.com/Digi-Key/digikey-kicad-library/zip/1fcb4bbab6e0cfa7658bcccbf4e7f6c643c3b9e5

mj1911, 2023-03-20:
Do note that KiCAD is at version 7.0.1 currently.  I for one, am all-for
stable versions of software; however 4.0.7. is light-years behind the
current builds of KiCAD.  KiCAD is updating rapidly and has changed much
since 4.0.7; following these old tutorials and versions is leaving you
significantly behind major improvements.