Digi-Key/digikey-kicad-library #139: There needs to be a better release process

wirehead, 2020-08-01:
I have to fab a second set of PCBs because of an issue that's already been fixed in this library.

The part in question is 10103594-0001LF.  Bug #76 was filed in 2018 noting that the size and shape of the prongs was wrong.  It was "fixed" in #126 a year later, but not really fixed.

I was trying to solder up a PCB and I discovered that the part wouldn't fit.  Got out my micrometer, discover that it's a bit off.  And then I see in the git log that the problem has been fixed and wonder why I'm not seeing the fixed footprint and that's because it's been "fixed" but whatever process you've got that copies things around the repository hasn't been run in 12 months.

It really sucks to have a bad batch of PCBs because of a bug that was almost fixed 10 months ago.

jacky4566, 2020-08-01:
Did you import the fixed library to your kicad?
AFAIK there is no process to update your local library automatically. You need to manually copy this master to your library location. 

I like to use Github desktop for automatically updating my Github libraries.

wirehead, 2020-08-01:
The problem here is that the fix is only in the `src/` directory (which KiCad isn't configured to use if I follow the provided instructions) and never made it over to the `digikey-footprints.pretty/` directory, not that I don't know how to use git.

bombledmonk, 2020-08-01:
I apologize, this is my fault in not getting the CI stuff up and running.  I'll see if I can't get this to updated and the CI stuff to work yet this week.

bschwind, 2021-01-27:
I just got bit by this as well. I've since manually copied the footprint over from the `src/` directory to `digikey-footprints.pretty/` as @wirehead has done, but should a PR be opened to move everything over? I imagine it shouldn't be much more than a simple script to copy the footprints over, but maybe there's more to it.