Digi-Key/digikey-kicad-library #157: Submit library to PCM metadata repo

OmegaRogue, 2021-11-24:
Submitting this library to the new Plugin And Content Manager would make it easier for users to install and find this library. The PCM is enabled in the current nightlies and will most likely be a part of kicad 6

Documentation for this is available in:
https://dev-docs.kicad.org/en/addons
The Metadata repo is located on:
https://gitlab.com/kicad/addons/metadata

yaqwsx, 2022-01-02:
Please, do so! PCM is the easiest way for BFU to use your library and you also will be properly credited!

bombledmonk, 2022-02-02:
Thanks for the suggestion, If I'm able to find the resources to get the library up to snuff this'll be on the list of things to happen.

OmegaRogue, 2022-02-03:
I've forked this and made a quick manifest and release of the current version for my own unofficial kicad pcm repo. I can help with getting stuff to kicad 6 and after that with packaging the library

OmegaRogue, 2022-02-03:
I was working on porting it, but found that the dcm files in src are in the gitignore, and the ones in the release are not complete