Digi-Key/digikey-kicad-library #165: Library Status

OmegaRogue, 2022-01-21:
is this library still maintained?

bombledmonk, 2022-02-02:
I added an advisory to the top of the readme warning that this is not currently maintained.  I'm still looking at options on how it might be updated again, but at this point it's well past due to have an overt warning.

OmegaRogue, 2022-02-14:
> I added an advisory to the top of the readme warning that this is not currently maintained.  I'm still looking at options on how it might be updated again, but at this point it's well past due to have an overt warning.

As I've stated in #157, I would be willing to update the library to KiCad 6, but currently the dcm files for the symbols in src are on the gitignore and missing from the repo, and it looks like there is an out of tree workflow for creating the final library files being used